helium/longfi-arduino #26: After EV_REJOIN_FAILED the board seems to lock up with no re-join attempts?

bubba198, 2020-12-25:
Hi,

The example code works very well. Joins Helium and pushed payload just fine. That is until after EV_REJOIN_FAILED message is received and then the board seems to drop-off Hilum and locks up with no re-join attempts thereafter?

Only reset gets the board out of this state. With mobile setup EV_LINK_DEAD is not uncommon as reception does change when the node moves from the range of one hot spot to another but it looks like that once EV_LINK_DEAD occurs re-join is not attempted and the node is basically dead. Here's the sequence of events which leads to this condition:

```
43201269: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
43671127: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
44140952: EV_TXCOMPLETE (includes waiting for RX windows)
44140974: EV_LINK_DEAD
OP_TXRXPEND, not sending
44695777: EV_REJOIN_FAILED

```
Any possible remedy to the problem?

Thank you
Boyan

kent-williams, 2021-01-15:
Hi @bubba198, 

Very sorry this got no response sooner!

Try increasing the `LINK_CHECK_DONE` (default is 24) in the `lmic.h` file found here at line 36:
```
linux: /home/{user}/Arduino/libraries/IBM_LMIC_framework/src/lmic/lmic.h
windows: Documents/Arduino/libraries/IBM_LMIC_framework/src/lmic/lmic.h
mac os: Documents/Arduino/libraries/IBM_LMIC_framework/src/lmic/lmic.h
```

kent-williams, 2021-11-22:
Closing this as there has been no response in some time.

closed 2021-11-22 by kent-williams