helium/longfi-arduino #29: Feature Request: Please Update Library to include The New Arduino Portenta Vision Shield LoRa version

hpssjellis, 2021-03-08:
Please Update the LongFi Library to include The New Arduino PortentaH7 Vision Shield LoRa version.

https://store.arduino.cc/usa/portenta-vision-shield-lora

Clips onto the https://store.arduino.cc/usa/portenta-h7 dual core 160 pin $100 USD board.

Makes machine learning possible on an edge device, perfect for the Helium network. I would test code for you but presently 40 km away from a Gateway, and new Gateways are not shipping very fast.

illperipherals, 2021-03-08:
https://store.arduino.cc/usa/portenta-vision-shield-lora

hpssjellis, 2021-03-30:
Any headway on the Portenta with LoRa Vision Shield @kent-williams ? I know a new version of the [Arduino mbed core](https://github.com/arduino/ArduinoCore-mbed)  (hopefully out soon) will have some improvements that might make LoRaWan easier to work with. Also the arm MBED API has some [LoRa](https://os.mbed.com/docs/mbed-os/v6.9/apis/loraradio.html) and [LoRaWan](https://os.mbed.com/docs/mbed-os/v6.9/apis/lorawan-apis.html) examples.

Presently my raw example [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/research/LoRaWan/helium-connected03.ino) does connect to Helium, but my experience is a bit lacking for the next steps of data sending and encryption. (Also not having a nearby gateway is a pain). Anyway,  I am willing to help in any way that I can.