helium/longfi-arduino #33: Different port than ST B-L072Z-LRWAN1 guide

JoeKlimcak, 2021-05-24:
Hello, I've recently purchased an ST B-L072Z-LRWAN1. Followed the guides on [Helium](https://docs.helium.com/use-the-network/devices/development/stmicroelectronics/st-b-l072z-lrwan1/arduino/), and in your [README](https://github.com/helium/longfi-arduino/tree/master/ST-B-L072Z-LRWAN1/longfi-us915). 

I have a problem in the last step. When it says: "_Navigate to (Tools > Port: COM# or /dev/ttyACM# or usbmodem) depending on whether you are on Windows, Mac, or Linux_.", I only see "/dev/cu.Bluetooth-Incoming-Port" in Tools > Port. I'm using a Mac.

Can you please tell me what I'm doing wrong?

leroyle, 2021-05-27:
Joe, have you commented in the Sensor-dev channel on Discord? Might need a driver, I'm not a Mac-spert.

JoeKlimcak, 2021-06-05:
Switching cables indeed helped (or in my case, stop using adapters). Thanks

closed 2021-06-05 by JoeKlimcak