helium/longfi-arduino #35: Unable to send data to HELIUM network

saadi-tech, 2021-09-10:
Hi, I am using a TTGO - T-beam V1.0 module. I am using EU868 frequency, and I created a device on Helium Console.
I followed all the steps given in the repository like editing the configuration file to LMIC, then updated the credentials and then ran the code.
The code was giving "MSG QUEUED" again and again, (ocassionally msg discarded too)
Please help me out what could have been gone wrong. Any hints for the solution or any method to check even if the LoRa module is working fine or not.

Please respond ASAP. Thank you so much.

saadi-tech, 2021-09-10:
One more thing please, do I need to use Device name and ID from helium console anywhere? I did not use them at any of the required steps.

kent-williams, 2021-09-15:
Hi @saadi-tech, you only need to use dev_eui, app_eui, and app_key from Console. Would you be able to share the serial debug output from your device after reset please.

saadi-tech, 2021-09-15:
Hi @kent-williams  thank you for getting back to me. Sure, I will share the serial debug screen in a while with you. 
The strange thing now is, that I can send data to "TheThingsNetwork" now, but still no activity on helium. :-P

I will share more information here with you

Thank you

kent-williams, 2021-09-15:
@saadi-tech Do you see anything in the Console event log, or nothing at all?

saadi-tech, 2021-09-19:
Hi @kent-williams , sorry I could not respond earlier. I have been out of the HELIUM coverage area from 4 days. I will get back to you once I get there and do some tests again. 
Thanks! :)

mauricecyril, 2021-09-20:
Hi I'm getting the same error. The request is sent to the network but the response keeps on getting returned invalid.
On the console I've setup the following decode function: https://github.com/helium/longfi-arduino/blob/master/TTGO-TBeam-Tracker/console-decoders/mapper_decoder.js

Serial output:

saadi-tech, 2021-10-07:
@kent-williams , Hi I tried it out right now, (the code I am using works with TTN perfectly, but not working with the Helium.) (I did use the credentials from Helium device)
Here is the serial output. (I have enabled debug msg display from LMC library):

[image]

kent-williams, 2021-11-22:
@saadi-tech is this working for you now?

saadi-tech, 2021-11-22:
@kent-williams surprisingly yes!!
I mean its ghost activity. One day it wasn't working at all ... Nd now its working perfectly. And I didn't change anything 🤣

kent-williams, 2021-11-22:
That's great to hear. The network has experienced a few outages over the last month, it's best to subscribe to https://status.helium.com/, so you know if they occur. Do you mind closing this issue if that is all please!

saadi-tech, 2021-11-22:
Sure @kent-williams . Thank you for assistance

saadi-tech, 2021-11-22:
Just got it working... Don't know how

closed 2021-11-22 by saadi-tech