helium/longfi-arduino #42: Get stuck showing "Joining". What configuration with region as "CN470"?

kirkang, 2021-11-19:
Get stuck showing "Joining". Some questions about configuration with region as "CN470".
1.  Directly add code in ESP32_LoRaWAN.cpp like below?
`#ifdef REGION_CN470`
`#include "region/RegionCN470.h"`
`#endif`
2. Should modify "RegionCN470.h"?
3. What's the LORAWAN_DEFAULT_DATARATE setting?
4. Anything others to solve stuck in showing "Joining"

kent-williams, 2021-11-19:
Hi @kirkang, which specific example are you trying to use? Are you in a CN470 region with a CN470 hotspot in range?

kirkang, 2021-11-20:
Hi @kent-williams, I am following this guide https://docs.helium.com/use-the-network/devices/development/heltec/wifi-lora-32-v2/arduino/. Yes I am in the CN470 region, it shows there is one hotspot in the hex I am in from https://explorer.helium.com/

kent-williams, 2021-11-22:
For starters, you may not be in range of the hotspot you are seeing in explorer for a successful join. Do you own a hotspot that you could test with in close proximity to begin with?

kent-williams, 2021-12-15:
Closing this, as there has been no response for some time.

closed 2021-12-15 by kent-williams