helium/longfi-arduino #44: No output on Serial or in Helium Console after uploading

ChrisX33, 2022-01-10:
Hi, I have compiled and uploaded the sketch to my Adafruit Feather M0 board but I get no output at all. It's as if nothing is happening. Arduino IDE gives me the following message:

```
Sketch uses 31820 bytes (12%) of program storage space. Maximum is 262144 bytes.
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 1.045 seconds

Write 32116 bytes to flash (502 pages)
[==============================] 100% (502/502 pages)
done in 0.248 seconds

Verify 32116 bytes of flash with checksum.
Verify successful
done in 0.027 seconds
CPU reset.
```

So all seems ok but I am not getting anything.

ChrisX33, 2022-01-10:
I am using this [tutorial](https://docs.helium.com/use-the-network/devices/development/adafruit/adafruit-feather-m0-rfm95/adafruitio/) minus the DHT22 and instead using the Hello World script.

robomaniac, 2022-05-26:
@ChrisX33 have a look if you use
`SerialUSB.begin(9600);`
or
`Serial.begin(9600);`