helium/longfi-arduino #47: How to increase appDataSize beyond 4

ksaye, 2022-03-22:
In reading [Line 101](https://github.com/helium/longfi-arduino/blob/master/Heltec-WiFi-LoRa-32-V2/longfi-us915/longfi-us915.ino#L101), it would appear that we can increase the appData up to 64.

If I increase it beyond 4, I get the payload size of 0.

```
static void prepareTxFrame( uint8_t port )
{
    appDataSize = 16;//AppDataSize max value is 64
    for(int counter = 0;counter < 16;counter++) {
       //statements block will executed 10 times
       appData[counter] = 0x00;
    }
}
```

[image]

What am I missing here?

kent-williams, 2022-03-22:
I believe you need to increase the data rate, see instructions [here](https://github.com/helium/longfi-arduino/tree/master/Heltec-WiFi-LoRa-32-V2/longfi-us915#required-change-to-default-datarate).

ksaye, 2022-03-23:
I tried all 4 data rates and even ADR, none helped.

After commenting out line 69 in: https://github.com/HelTecAutomation/ESP32_LoRaWAN/blob/master/src/ESP32_LoRaWAN.cpp#L69, it was able to send a much larger packet.  I guess I need to research LoRaMacQueryTxPossible()

[image]

closed 2022-03-23 by ksaye

ksaye, 2022-03-23:
FYI. I was even able to send a 128 bit packet, shown below.  Nice to know if I need to send a bit larger packet that it works.

[image]