helium/longfi-arduino #50: Heltec WiFi LoRa 32 V2/us915. Get stuck showing "Joining"

ztjona, 2022-06-07:
Hi I want to send some data using a Heltec WiFi LoRa 32 V2.

I follow the guide in:
https://docs.helium.com/use-the-network/devices/development/heltec/wifi-lora-32-v2/arduino/

I have a hotspot that is in 1 floor above where I am testing the board. 

I do not know why, but it can not join. 
I modified the build flags to show more info. This is the log I obtained.

```
Init

LoRaWAN US915 Class A start!

JOIN
joining...
TX on freq 904600000 Hz at DR 4
SLEEP
DIO0:TX Done
RX on freq 923900000 Hz at DR 13
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
TX on freq 905100000 Hz at DR 0
DIO0:TX Done
RX on freq 926900000 Hz at DR 10
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
join failed, rejoin at 30000 ms later
Deep Sleep until Next TxPacket:28900 ms
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Init

LoRaWAN US915 Class A start!

JOIN
joining...
TX on freq 904600000 Hz at DR 4
SLEEP
DIO0:TX Done
RX on freq 923900000 Hz at DR 13
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
TX on freq 905100000 Hz at DR 0
DIO0:TX Done
RX on freq 926900000 Hz at DR 10
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
join failed, rejoin at 30000 ms later
Deep Sleep until Next TxPacket:28900 ms
ESP32 MCU init...
ESP32 MCU inited OK!
Init

LoRaWAN US915 Class A start!

JOIN
joining...
TX on freq 904600000 Hz at DR 4
SLEEP
DIO0:TX Done
RX on freq 923900000 Hz at DR 13
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
TX on freq 904300000 Hz at DR 0
DIO0:TX Done
RX on freq 924500000 Hz at DR 10
DIO1:RX Timeout
RX on freq 923300000 Hz at DR 8
DIO1:RX Timeout
join failed, rejoin at 30000 ms later
Deep Sleep until Next TxPacket:28900 ms
```

I tried data rate 0 and 3:
`#define LORAWAN_DEFAULT_DATARATE DR_3`

I also changed the mask to:
`uint16_t userChannelsMask[6] = {0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};`

Sometimes it also reboots, and I do not know why either.

kent-williams, 2022-06-08:
Just to be sure, are you located in the proper region(US) you are trying to join on? Make sure to double check your devEUI, appEUI, and appKey.

ztjona, 2022-06-08:
The region is ok, my hotspot is US915 and it works fine in the helium network.
The devEUI, appEUI, and appKey I just copied from a new device I created in Helium console. I double check it, it is fine.