helium/longfi-arduino #53: Heltec WiFi LoRa 32 V3 support

miguelmoraleda, 2023-01-09:
Hello,

I have been trying to run [this ](https://github.com/helium/longfi-arduino/blob/master/Heltec-WiFi-LoRa-32-V2/longfi-us915/longfi-us915.ino) with the latest V3 version but its giving me compilations errors. This may not be an issue as im totally new to this devices/coding platform. Could someone confirm it is possible to run with v3? 

The error I get is:

`Compilation error: 'RST_LoRa' was not declared in this scope`

I tried bypassing this errors by giving the specific pin numbers but then ended up having other errors so I wanted to ask here.

closed 2023-01-18 by miguelmoraleda

gonesurfing, 2023-01-22:
I'm running into the same error. How did you resolve it?

miguelmoraleda, 2023-01-22:
I haven't been able to resolve this yet but maybe getting closer to solve my problem. 

I'm using https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series

And found https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/libraries/Heltec-Example/examples/LoRaWAN/LoRaWan_license/LoRaWan_license.ino .... I was trying this with no luck, please let me now if you find a way to make it work 🙏 

I was able to compile that one but it doesn't join the network.

gonesurfing, 2023-01-22:
I forked the repo, and got a start on a V3 sketch.

https://github.com/gonesurfing/longfi-arduino

see
https://github.com/gonesurfing/longfi-arduino/tree/master/Heltec-WiFi-LoRa-32-V3/longfi-us915

Key changes:
-New new 0.0.7 board file includes the new libraries. No extra library download is needed apparently.
-I don't see where this new library takes a license. Maybe it's not needed anymore?

It compiles/boots, but I haven't had any luck joining yet. I may have to drive around and find a network. Please test if you're near the helium network!

miguelmoraleda, 2023-01-22:
without a license? or did you end up adding the license as in the example I shared before? (https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/libraries/Heltec-Example/examples/LoRaWAN/LoRaWan_license/LoRaWan_license.ino)

I'll give it a try but I think it is exactly what I have already.

gonesurfing, 2023-01-23:
+ADR=0
+IsTxConfirmed=0
+AppPort=2
+DutyCycle=15000
+ConfirmedNbTrials=8
+ChMask=0000000200000000000000FF
+DevEui=
+AppEui=
+AppKey=
+NwkSKey=
+AppSKey=

+DevAddr=007E6AE1(For ABP Mode)
joining...
joined

unconfirmed uplink sending ...

miguelmoraleda, 2023-01-23:
Mine get stay on joining... and then I have join failed every few mins. 

What region are you using ? us915? what did you do on the helium console? just created a new device ?

gonesurfing, 2023-01-23:
I just created a new device, and copied the three hex codes over to the Arduino sketch.

I'm using us915. I also modified the channel mask per the helium wiki page. That may have sped things up.

`uint16_t userChannelsMask[6]={ 0xFF00,0x0000,0x0000,0x0000,0x0002,0x0000 };`

miguelmoraleda, 2023-01-23:
Im doing exactly the same.. I tried your code with no luck :( so weird... how long does it take to join?

gonesurfing, 2023-01-23:
It was very quick, once I was an in area with a hotspot. Maybe 10 seconds.  I am assuming you are in an area that has coverage? The map is a little misleading... Some of the nodes are inactive but still show up as providing coverage.

miguelmoraleda, 2023-01-23:
I have a hotspot at home, in theory I should have plenty of coverage. I went for a walk as well by no luck. I’ll keep trying, I’m glad it did work for you. At least I’m on track to success I guess Sent from my iPhone

gonesurfing, 2023-01-24:
I think I may have declared success too early.  The only data the board will transmit is the join request. The actual payload is not transferring.

I think it makes sense to either re-open this ticket or start a new one. I'm sure there are more people struggling with this V3 mess.

miguelmoraleda, 2023-01-24:
I'm still struggling... I started a thread on Heltec's forum.

http://community.heltec.cn/t/heltec-wifi-lora-32-v3-helium-network/12329

gonesurfing, 2023-01-24:
Thanks. I’ll follow that thread. I don’t understand when the license is required. Is it the library or the Lora chip that needs the license?